Safecast/bGeigieZen #37: File naming scheme for Drive and Journal

louisbertrand, 2023-08-13:
The file names should encode essential information to show status at a glance. The Core and Core2 API permit writing file names longer than the old DOS 8.3 constraint, and also create directories and subdirectories.
The information for Drive:
date-time started (first record written)
has it been submitted?
** what else? **

louisbertrand, 2023-08-21:
Existing bGeigie Nano file naming is
UUUUMMDD.LOG
where UUUU stands for the 4-digit device ID, and MMDD is 2-digit month and 2-digit day of the month. The year cannot be encoded in the file name, so logging on the same month/day in multiple years will append to the same file. A workaround is to mount the uSD card on a PC and archive the file after it has been submitted to the API. Manually move the latest files to a separate folder for each year.

louisbertrand, 2023-08-21:
Open questions:
If the file name must include date and time (e.g. ISO standard YYYY-MM-DDTHH:MM:SSZ), can a journal file be opened and named before getting a GPS fix? 
Create a separate folder for each year?
What status needs to be encoded? New, Submitted, what else?

robouden, 2023-08-21:
> Existing bGeigie Nano file naming is UUUUMMDD.LOG where UUUU stands for the 4-digit device ID, and MMDD is 2-digit month and 2-digit day of the month. The year cannot be encoded in the file name, so [...]

The way that Robin worded around that is by creating year folders in the Zen..

closed 2024-03-06 by robouden