Safecast/bGeigieZen #38: How big can a QR code be? Size? # of characters?

louisbertrand, 2023-08-13:
Establish what the limitations are for the size on-screen and number of characters for the M5Stack API QR code generator.
Also look into a URL shortener that does not use an intermediary site.

louisbertrand, 2023-08-23:
The longest string we should be able to encode as a QR is 134 ASCII characters. It would be 195 if using only alphanumeric mode characters 0-9, A-Z, { $%*+-./:} (includes the space character), but the base URL definition QR_CODE_URL_BASE="http://tt.safecast.org/id/geigiecast:" in user_config.h uses lowercase characters .
Based on the M5Stack library files M5Display.{h,cpp} and utility/qrcode.{h,c}, the API generates a QR code version=6 with low ECC. The version number gives the size. The M5Stack API gives a 220px square (height of the display is 240px). The library reference by author Richard Moore (https://github.com/ricmoo/qrcode/) gives the limits for versions and characters sets used.

robouden, 2023-08-23:
We can make a URL shortener.. https://www.twilio.com/blog/build-a-url-shortener-with-csharp-dotnet-and-redis

closed 2024-03-06 by robouden