Safecast/bGeigieZen #39: Can date-time be known before GPS ready?

louisbertrand, 2023-08-13:
It would be very convenient to begin journalling as soon as the system is started, but without date-time, it's hard to write reliable information. 
Is there a real-time clock on the M5Stack Core and Core2?
Can the millis() be relied upon if the system sleeps? Is there an indication that a boot has occurred (cold boot, warm boot)?

louisbertrand, 2023-08-21:
The Core2 includes a RTC (NXP BM8563) with Arduino driver https://github.com/tanakamasayuki/I2C_BM8563
Unfortunately, the original Core does not.

robouden, 2023-08-21:
I checked the RTC is on the main board, not on the extension board:

From the M5Stack docs: 
"There is a small expansion board on the back of the base with a 6-axis IMU sensor and microphone. There is a small expansion board on the back of the base with a 6-axis IMU sensor and microphone. "

louisbertrand, 2023-10-07:
For journalling purposes only, I propose to add an item in local storage for "last known date-time", to be updated at the same time as last_longitude and last_latitude. This last known date-time can be used as the file name for a new journal file on a cold start (no GNSS time known, and no realtime clock IC available.
The last known date-time can be kept in class LocalStorage (local_storage.h).
When the system boots, the journal file is created with a file name based on the last known date-time:
YYYY-MM-DD_HH-MM-SS_TEMP_JRNLTXT
Entries are timestamped with the millis() value.
When the correct time becomes available (GNSS), the journal file should be renamed. The renaming should be noted in the journal file so that older entries can be accurately back-dated based the milliseconds timestamps, relative to when the file was renamed.
The last known time is strictly for internal purposes; it would not be visible to the end user or logged to the map database.

BeakeS, 2023-10-07:
How can I purchase one of these kits?  The website store just hangs when I
try to checkout.

I could help with code development if I had the hardware.

Michael Beakes

louisbertrand, 2023-10-07:
Hi @BeakeS, the kits are still under development, with only a few prototypes available. Until I got mine, I just used the individual components on breadboards (M5Stack Core and Core2, BN-880 GPS). Without a GM tube, I just used an Arduino to provide pulses.
If you'd like to follow up, the best contact person is lead developer @robouden (Rob Oudendijk).

robouden, 2023-10-07:
@BeakeS

If you want, you can put in a custom order. Currently, we are just taken in orders. Not selling yet. Production is a work in progress. If you want to help with software development, please join our weekly meeting at:

Join Zoom Meeting
[https://us04web.zoom.us/j/3104600384?pwd=M2tZUGNWQll6aTJEMnByN1YxWUxrdz09](https://www.google.com/url?q=https://us04web.zoom.us/j/3104600384?pwd%3DM2tZUGNWQll6aTJEMnByN1YxWUxrdz09&sa=D&source=calendar&usd=2&usg=AOvVaw0VgS3AhOIiQhz7jdKR2oJE)

Meeting ID: 310 460 0384
Passcode: 9ZjWD6

Schedule is below:
[image]

robouden, 2023-10-07:
> I checked the RTC is on the main board, not on the extension board:
> 
> From the M5Stack docs: "There is a small expansion board on the back of the base with a 6-axis IMU sensor and microphone. There is a small expansion board on the back of the base with a 6-axis IMU sensor and microphone. "

Great find. Did you check if it works?
!!!!!

louisbertrand, 2023-10-07:
Yes, the RTC works well. There is a "not initialized" bit that we can query, but it starts keeping time immediately even if the starting point is random after initial power up.

robouden, 2023-10-07:
And if the power is off, still on boot it has the correct time?

Regards,
Rob

louisbertrand, 2023-10-07:
Yup, battery backed up, but I need to confirm. I've had it powered off for a while, good opportunity to check.

robouden, 2023-10-07:
Seems you figured it out already

louisbertrand, 2023-10-07:
Yes, but also confirm on the actual prototype, with the external battery and charger.

closed 2023-10-07 by robouden

robouden, 2023-10-09:
BeakeS,

If you send us your email, I can invite you for the Slack channel of bGeigieZen for software and hardware discussions.

Regards,
Rob Oudendijk

robouden, 2023-10-09:
BeakeS,  

Interesting email address.
Noticed you have 0 repos on GitHub, are you new to GitHub?

 

Regards,
Rob Oudendijk

BeakeS, 2023-10-09:
Not new to github but not an expert either.  Had a different account/email
when I was a mentor for high school robotics teams.

Since I retired in 2021, I've been working on a lot of
microcontroller-based projects - mostly networked home sensor/control for
optimizing multi-zone hvac, water well monitoring, and security.  My son
has been bugging me for some time to push all my code off my linux box and
onto github.  No time like the present.

Michael

robouden, 2023-10-09:
BeakeS,

Thanks for the reply.
Better keep all your code on GitHub if you work in a team. So easy to roll
back, branch etc.

You want me to add you to the Safecast Slack channel?

Regards,
Rob Oudendijk

BeakeS, 2023-10-09:
Yes, please do.

Thanks,
Michael

robouden, 2023-10-09:
BeakeS,

Still need your email address.

Rob

BeakeS, 2023-10-09:
***@***.***

(I sent it when you first asked but changed the subject line.)

Thanks.

robouden, 2023-10-10:
Can you resend it to my email address below, or maybe better with Signal or
Keybase?

Regards,
Rob Oudendijk
https://yr-design.biz <http://yr-design.biz/>   https://oudendijk.biz
<http://oudendijk.biz/>  https://about.me/robouden
<http://about.me/robouden>tel +81 80-22605966  Skype: robouden  Facebook:
robouden <http://on.fb.me/QeKw2P> linkedin:robouden
<https://www.linkedin.com/in/roboudendijk>