Safecast/bGeigieZen #40: Cold and warm start of GPS

robouden, 2023-08-22:
After Azby did some testing I thought maybe his Zen did not get a first good lock and stayed in “cold start” mode.

Did we implement as on the Nano the following?:

For the bGeigieNano we use 
#define PMTK_HOT_START "$PMTK101*32"
#define PMTK_COLD_START "$PMTK104*37"

louisbertrand, 2023-08-23:
I'm confused... Are these commands to send to the device? Or status sent by the device as it starts receiving GPS signals? 
The TinyGPS++ API might offer that kind of detail.
Normally we would rely on the return values from isValid() and isUpdated(). However, you can define your own custom object `TinyGPSCustom` giving the sentence name and the field to watch. See the section "Differences from TinyGPS" in http://arduiniana.org/libraries/tinygpsplus/
I am going to try something like this, but I'm not sure what the parser will do with term number 0 (i.e. there are no data fields in this sentence, just the header followed by the checksum.
`TinyGPSCustom warmStart(gps, "PMTK102", 0);`

robouden, 2023-08-23:
Let us know the results.  The GPS should lock in a few seconds instead of 30 seconds or longer

louisbertrand, 2023-10-07:
After digging into how GNSS (Global Navigation Satellite Systems) work in general, and the u-blox M8 module in particular. we are stuck with long cold start times but shorter restart times. 
There are multiple GNSS systems out there, and some augmentation methods to improve fix quality and acquisition times. GPS (NAVSTAR, USA) is the oldest and best known, but GLONASS (Russia) and Galileo (EU) are also available globally. In Asia, regional systems are available from China, Japan and India. Regional systems use specific orbits to keep satellites visible over their designated region. Geostationary satellites and terrestrial broadcasts linked to the global systems   improve fix acquisition time and quality. 
Incidentally, smartphones can get their fix quickly because they get their ephemeris from the cellular phone provider.
In a cold start, the receiver has to wait until it has the almanac (orbit parameters) from the chosen constellation, up to 32 satellites, transmitted at 50bits/sec (yes, slow), then get precise ephemeris from each satellite in view. This can take up to 12 minutes for older receivers. Newer receivers read multiple channels concurrently and use augmentation systems for faster acquisition times, typically less than 4 minutes. The ability to use assistance systems depends on the receiver's location on the globe.
For a warm start, the receiver uses almanac stored in battery backed-up RAM (BBR) as "good enough for rock-n-roll" until the actual data catches up. The BBR data is good for up to 4 hours. After that, the time to a reliable fix is longer, though not as long as a cold start.
**Conclusion:** If the satellite almanac is unknown or stale, we can't get a fix any faster than what the receiver can get from the system. This would have to be pointed out in the user documentation so that a new user understands that the device is working correctly even if it doesn't get a fix immediately. 
However, if we mistrust the fix, we have the option of forcing a cold start to discard discard the current almanac and ephemeris and reload from the system.

louisbertrand, 2023-10-07:
Correcting a misconception about warm-start vs. cold-start. Forcing a cold start is only relevant if we need to reload satellite almanac from scratch. Forcing a warm start only makes sense if the almanac is fresh (less than a few hours old).

closed 2023-10-07 by louisbertrand