Safecast/bGeigieZen #72: Define reqs for power mgmt v4.x boards

louisbertrand, 2024-05-19:
Define the requirements and follow-up design goals for the power management on v4.x boards: powering the device, charging the battery, power-on/power-off, protection.
DEFINITIONS (loosely based on AXP2101 terminology).
VBUS = 5V from USB charger
VSYS = System voltage for AXP2101
VBAT = Battery voltage
Other voltages mentioned are for internal use in CORE2 V1.1

GIVENS:
M5Stack CORE2 v1.1. The original CORE2 no longer available because the AXP192 PMIC was obsoleted by the manufacturer.
Single cell Li-Ion battery 18650 3.7V 18mm x 65 mm cylindrical form factor, capacities range up to 3400mAh 

1- MUST HAVE
Reverse battery protection: against inserting battery in wrong polarity.
Over-voltage protection on VBAT. 
Over-discharge protection
Over-charge protection for the battery
Charging without connecting USB cable: avoid connector point of failure.

2 - IMPORTANT TO HAVE
Low component count: cost, build complexity and, well, Zen.
No SMT soldering: easy for inexperienced kit builders (power device like DPAK or D2PAK would be OK).
Extended operating time between recharges; what is the target?
Remove power switch that was on v3.x boards
Charging without opening case, i.e. Qi charger

3 - NICE TO HAVE
Battery fast charge I(CH) > 300mA (AXP2101 default).
Control over power off: firmware read of CORE2 power button and control of power on/off through AXP2101 control register.

robouden, 2024-05-19:
The AXP2101 max charging is 100mA!!!

robouden, 2024-05-24:
Nice to have fast charging, still needs to be done..

closed 2025-01-14 by robouden