Safecast/bGeigieZen #73: Cannot connect to local WiFi in Real-time mode

louisbertrand, 2024-05-20:
I cannot connect in real-time mode to the hotspot or to the local WiFi network.

Hardware CORE2 v1.1, running the updated firmware as of today May 20 (`23f5a8f8c1c14cd66cff0aeb73578ed245a44c29`)

The serial port log when entering Real-time mode:
```
New screen entered: Real-time
E (185365) wifi_init_default: netstack cb reg failed with 12308
WiFi connector: Trying to connect to wifi (Pixel_2172:zzzzzzzz)...
```
But the settings from the built-in AP and web server show the network name and password are correct.
[image]

I get the same when trying to connect to the local WiFi:
```
New screen entered: Real-time
E (832090) wifi_init_default: netstack cb reg failed with 12308
WiFi connector: Trying to connect to wifi (cavedweller:zzzzzzzz)...
```

Claypuppet, 2024-05-20:
The ESP32 can only connect to 2.4GHz networks, can you confirm the networks you tried to connect to are 2.4 and not 5GHz? Should probably be added in the manual too.

louisbertrand, 2024-05-20:
My phone only offers 5GHz??? That seems to be the problem. I'm closing it with a note to add to the documentation.

closed 2024-05-20 by louisbertrand