Safecast/bGeigieZen #77: Password length limitation

eutill, 2024-09-09:
Hello,
thanks for the development of the bGeigieZen. We encountered a small issue when trying to connect to a local WiFi network (that is, to an existing Access Point).
After configuring the WiFi password in the configuration file on the SD card, the bGeigieZen couldn't connect to the Wifi network when selecting "Start on local" in the Config menu.
After setting up an Access Point on the bGeigieZen, we connected to the local server and tried to change the Wifi password in the Configuration Settings, but after clicking "Save", it wouldn't accept our password...
After narrowing down the possible problem sources, we found that the problem seems to be the password length. (Ours is 35 characters long.)
We would kindly like to know whether this is a real limitiation of the device or anything else.
Firmware version used: v3.2.3
Thank you!
Leo and Jean-François

robouden, 2024-09-10:
Leo,

Thanks for informing us. 
We will look into the issue.

Rob Oudendijk

JFB-ezy, 2024-09-10:
Thanks for looking into, Rob, this will be appreciated.
Have a good day
Jean-François

JFB-ezy, 2024-09-11:
Hello, after rechecking the firmare version burnt, it was the V3.2.5 and not the V3.2.3 as indicated previously!
Sorry for the mismatch.
Regards
Jean-François

robouden, 2024-10-02:
The latest update for the bGeigieZen , V3.5.6 has 64 characters enabled. And also many optimizations for the config server.
Please check it out. [https://bgeigiezen.safecast.jp/news-updates/](https://bgeigiezen.safecast.jp/news-updates/) 

Regards
Rob Oudendijk

eutill, 2024-10-02:
Thanks for the info, I can see the changes made in commit f35e41c1885bfcdc2b8b7c4a652b7a03dac2425e.
Maybe @JFB-ezy can flash the latest firmware v3.2.6 onto his unit and verify that his device now connects to the network.
Regards
Leo

JFB-ezy, 2024-10-02:
Hello,
Thank you for this update. I loaded the new version v3.2.6 and tried to test if my WIFI network with 35 characters is able to accept the ZEN device.
First: I start the ZEN device and in setting tag the wifi credentials are the one from the SD previously tested with 10 characters.
Second: I load The datas from the SD card ( filename SAFEZEN.txt) and the old password with 10 characters is not updated while the name of the SSID is correctly updated. 
Third: checking if the Wifi from my home ( with 35 characters password) is able to accept the connection and refuses the connection.
fourth: The old network SSID with the 10 Characters is able to accept the ZEN connection, after reloading with the old SAFEZEN.txt file reloaded
Conclusion: it seems that loading from the SD card doesnot accept the 35 characters password.
Sorry for this bad news!
If you thing that I made a wrong sequence of test please advice!
Best regards,
JFB

JFB-ezy, 2024-10-07:
With the help of Rob, the only way, with the version 3.2.6, to use a long password ( more than 32 characters) is to set it via the access point bgeigiezen-XXXX network and modifying the local wifi password and saving it. The loading from the SD card doesnot work above 32 Characters!
It was rendered more difficult while using Firefox browser to perform the update of the password. I used Edge which is more stable for such an access!
Regards and thanks to ROB ( and Leo )
JFB

closed 2025-01-14 by robouden