Safecast/bGeigieZen #79: No data shown in android app using bGeigie Zen

ZaphodBeeblebroxIV, 2025-01-14:
When using the android app (v0.5.0) it is possible to connect to the bGeigie Zen (v3.2.8) and the "BT" symbol turns green when pressing the connection button in the app. However, there is no data or map in the app. I have tried the same app on two different android devices (android 9 and android 14) with the same result.

robouden, 2025-01-14:
Bjorn,

Thank you for creating the issue.

Regards
Rob Oudendijk

ZaphodBeeblebroxIV, 2025-01-15:
I think the problem is in the app. I investigated this a bit and I downloaded the app "Serial Bluetooth Terminal 1.48". I used the information the file "bluetooth_settings.h" to create a profile and the I was able to use this app to connect to my bGeigie. I received the following data (note: I have obfuscated my position).
_18:28:45.631 Connecting to bGeigie5082 ...
18:28:46.970 Connected
18:28:47.308 $BNRDD,5082,2025‑01‑15T17:28:49Z,39,5,12662,A,5828.1234,N,1538.1234,E,37.0,A,7,346*42$BNRDD,5082,2025‑01‑15T17:28:54Z,41,4,12666,A,5828.1234,N,1538.1234,E,36.7,A,8,344*49$BNRDD,5082,2025‑01‑15T17:28:59Z,44,7,12673,A,5828.1234,N,1538.1234,E,38.3,A,9,254*41
18:29:19.877 Disconnected_
I compared the above information with Appendix A in the manual and it seems to be valid data.
Hence, the bGeigie works properly sending data and problem seems to be in the app.

robouden, 2025-01-15:
Zaphhod,

Thanks for checking this. The creator of the android app and the IOS map (two different creators) are both currently not available or not willing to fix the issues. For both  apps, we have the code available. I tried to rebuild the app, but I could not get it to build. If you want to try yourself, please let me know and we can collaborate to get it working. Would be great to get it fixed!!!

Warm regards,
Rob Oudendijk

robouden, 2025-01-15:
Zaphod,

The app is on [https://github.com/Safecast/Drive-android](https://github.com/Safecast/Drive-android)
For building [https://elafargue.github.io/wizkers-doc/building/](https://elafargue.github.io/wizkers-doc/building/)

Regards,
Rob Oudendijk